fabianoriccardi/melody-player #1: Issue with servo when playing melody

playrobotics-alex, 2020-05-03:
Hi,

Thank you for the library!
I found an issue, when a sevo is connected and melody is played, the servo goes crazy at the time of melody play.
Is there something that can be done?

Thank you

fabiuz7, 2020-05-10:
Hi, sorry for the late reply. Which board are you using? ESP8266 or ESP32?

playrobotics-alex, 2020-05-11:
Hi, ESP32
I found somewhere that there is an optional channels parameter, so I  changed the channel and it worked!
MelodyPlayer player(buzzerPin,6);

fabiuz7, 2020-05-12:
I will insert it in the readme!

closed 2020-05-12 by fabiuz7

playrobotics-alex, 2020-05-24:
Thanks,
By the way would you consider to add your library to Arduino IDE library manager?
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

I am working on a tutorial that will mention your library and it would be great if people would be able to just simply use it from within the IDE.

Thanks again for your work!

fabiuz7, 2020-05-24:
Yeah, my intention was that, but in this period I had focused on other my libraries... However, I think that in a few days it will come to the turn of this! 

May you give me the link to your tutorial when ready?

playrobotics-alex, 2020-05-26:
Sure, basically it is a robot based on ESP32 with a passive buzzer.
Will share the link when ready!

fabiuz7, 2020-06-06:
[Request done](https://github.com/arduino/Arduino/issues/10318), I think it will be added in the next week

playrobotics-alex, 2020-06-06:
Amazing! Thank you very much!