fabianoriccardi/melody-player #2: RTTTL playback tempo

Sruell, 2020-08-04:
Thanks for the great library. I use it to play RTTTL ringtones on my ESP8266 alarm clock. The first melody is played directly from a hardcoded String, when the device is plugged in. This melody plays fine.
I'd also like to play melodies ad hoc that I send via MQTT (pubsubclient library) as RTTTL string to the device. This also works, except that the melody plays far too slow.

Do you have any idea what could cause such a slow playback?

flyingThuna, 2021-02-16:
I also enjoy this library, great work!
I noticed that rtttl's with bpm greater than 200 play back way too slow, e.g:
IceIceBaby:d=4,o=5,b=**225**:blablabla will play back in "slow motion", when I set b=200 it works fine except the melody sounds kinda off since it's the wrong bpm :/
In this case it wouldn't be too bad, but other melodys have way higher bpm

do you have any idea how to improve on that?
Thanks in advance!

fabiuz7, 2021-03-21:
Hi @flyingThuna, 
I have just pushed the update to overcome your problem. 
I hardcoded the bpm limitation to 200 (thinking 200 was enough), so your melody was played to the default value of 63bpm. I had extended the limitation to 300.

fabiuz7, 2021-03-21:
@Sruell more than 6 months are gone, I'm sorry. I hope you had solved your issue, if not let me know

flyingThuna, 2021-03-21:
Nice, thank you very much!

fabiuz7, 2021-03-29:
Just released the version containing the fix, it just be available in few hours on arduino library manager and platformio

closed 2021-03-29 by fabiuz7