fabianoriccardi/melody-player #3: Static tone on ESP32 when the MelodyPlayer object is instantiated

SPD13, 2021-03-05:
I have a static tone on ESP32, just by initiating the library:
`MelodyPlayer player(BUZZER_PIN);`
I found that I can get rid of this static tone by commenting `digitalWrite(pin, HIGH);` in the constructor and the `turnOff()` functions.
A fix looks like this:

`#ifdef ESP32
MelodyPlayer::MelodyPlayer(unsigned char pin, unsigned char pwmChannel):
    pin(pin), pwmChannel(pwmChannel), state(State::STOP), melodyState(nullptr) {
  pinMode(pin, OUTPUT);
  //digitalWrite(pin, HIGH);
};
#else
MelodyPlayer::MelodyPlayer(unsigned char pin):
    pin(pin), state(State::STOP), melodyState(nullptr) {
  pinMode(pin, OUTPUT);
  //digitalWrite(pin, HIGH);
};
#endif

void MelodyPlayer::haltPlay(){
  // Stop player, but do not reset the melodyState
  ticker.detach();
  turnOff();
}

void MelodyPlayer::turnOn(){
#ifdef ESP32
  const int resolution = 8;
  // 2000 is a frequency, it will be changed at the first play
  ledcSetup(pwmChannel, 2000, resolution);
  ledcAttachPin(pin, pwmChannel);
  ledcWrite(pwmChannel, 125);
#endif
}

void MelodyPlayer::turnOff() {
#ifdef ESP32
  ledcWrite(pwmChannel, 0);
  ledcDetachPin(pin);
#else
  // Remember that this will set LOW output
  noTone(pin);
#endif

  pinMode(pin, OUTPUT);
  //digitalWrite(pin, HIGH);
}
`
Is there a reason why the buzzer pin is set to HIGH?
I can send a pull request if this is a bug.

fabiuz7, 2021-03-14:
Hi,
thanks for commenting. I think that it was due to my specific version of buzzer, which are active low. Can you link your buzzer model or your wiring diagram?
I think that may be a limitation of my library, easily fixable.

SPD13, 2021-03-14:
Thanks for the reply, this is the model I use: https://www.adafruit.com/product/1536?gclid=CjwKCAiAhbeCBhBcEiwAkv2cY9sFcj1lCO-OGpGLQikY-sczwJnw5vaCQK7rvyjHzZKQTG755V50mBoCOG4QAvD_BwE
I confirm that by commenting the lines with "digitalWrite(pin, HIGH);" in "turnOn", "turnOff" and the constructor, the library works perfectly.
It could be added as a "define" option in a future version of the library.

fabiuz7, 2021-03-14:
Yes i could add a define or even better a parameter in the costructor. 

I have opened the link and I have a question for you: can you play music with that buzzer? Since it plays at fixed 2khz

fabiuz7, 2021-03-20:
I had released v2.2.0, now you should be able to select the off voltage level (default is HIGH, to avoid breaking the backward compatibility)

SPD13, 2021-03-23:
Yes, my model seems to don't have this 2khz restriction. The RTTTL melodies are working perfectly.
Thank you very much for the lib upgrade!

closed 2021-03-23 by fabiuz7