fabianoriccardi/melody-player #5: Error compiling for board SparkFun ESP32 Thing Plus.

mikulik86, 2022-03-04:
Hi,
I get an Error during compile. See the log:

```
Arduino: 1.8.19 (Windows 10), Board: "SparkFun ESP32 Thing Plus, 80MHz, Default (6.25MB APP/OTA/3.43MB SPIFFS), 921600, None"

In file included from C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody_player.h:4,

                 from C:\Users\Admin\Documents\Arduino\sketch_mar04a\sketch_mar04a.ino:8:

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:70: error: 'vector' is not a member of 'std'

   Melody(String title, unsigned short timeUnit, std::shared_ptr<std::vector<NoteDuration>> notes, bool automaticSilence):

                                                                      ^~~~~~

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:70: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:6:1:

+#include <vector>

 

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:70:

   Melody(String title, unsigned short timeUnit, std::shared_ptr<std::vector<NoteDuration>> notes, bool automaticSilence):

                                                                      ^~~~~~

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:70: error: 'vector' is not a member of 'std'

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:70: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:77: error: template argument 1 is invalid

   Melody(String title, unsigned short timeUnit, std::shared_ptr<std::vector<NoteDuration>> notes, bool automaticSilence):

                                                                             ^~~~~~~~~~~~

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:70: error: 'vector' is not a member of 'std'

   Melody(String title, unsigned short timeUnit, std::shared_ptr<std::vector<NoteDuration>> notes, bool automaticSilence):

                                                                      ^~~~~~

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:70: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:70: error: 'vector' is not a member of 'std'

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:70: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:89: error: template argument 1 is invalid

   Melody(String title, unsigned short timeUnit, std::shared_ptr<std::vector<NoteDuration>> notes, bool automaticSilence):

                                                                                         ^~

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:70: error: 'vector' is not a member of 'std'

   Melody(String title, unsigned short timeUnit, std::shared_ptr<std::vector<NoteDuration>> notes, bool automaticSilence):

                                                                      ^~~~~~

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:70: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:70: error: 'vector' is not a member of 'std'

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody.h:26:70: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?

[... 171 lines cut ...]

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody_factory.h:50:24: error: 'vector' is not a member of 'std'

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody_factory.h:50:24: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody_factory.h:50:31: error: template argument 1 is invalid

   std::shared_ptr<std::vector<NoteDuration>> notes;

                               ^~~~~~~~~~~~

C:\Users\Admin\Documents\Arduino\libraries\Melody_Player\src/melody_factory.h:50:43: error: expected unqualified-id before '>' token

   std::shared_ptr<std::vector<NoteDuration>> notes;

                                           ^~

exit status 1

Error compiling for board SparkFun ESP32 Thing Plus.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

fabianoriccardi, 2022-03-06:
Hi mikulik86,
thanks for the report, there was an issue with the recent versions of arduino cores, affecting both esp8266 and esp32. The latest commit https://github.com/fabianoriccardi/melody-player/commit/3c4f43530e9bf841c78382797990705cdc4efe6f fix the issue, I will release a new version soon.

closed 2022-03-06 by fabianoriccardi

fabianoriccardi, 2022-03-06:
Update released! It should appear in the next few hours on Arduino and Platformio registry, alternatively you can download the release from Github.

mikulik86, 2022-03-06:
Thanks for such a quick fix! ;)